Maritime-Robotics-Student-Society/sailing-robot #14: Clearify what happens to goal_heading during/after tacking

smaria, 2016-02-03:
When highlevel_control switches to tacking, what will be published in the goal_heading topic? How is it ensured that after the tack, the right goal_heading is published?

UnnamedMoose, 2016-02-04:
Personally I see 3 ways of defining this:
1. Tell the boat to get to a fixed GPS heading estimated based on the current true wind speed, making sure the tack is executed by fixing the rudder at a certain angle for a given amount of time (probably worked out through trial and error)
2. Telling the boat to cross the wind direction and then keep falling off the wind until apparent wind angle becomes about optimum for accelerating to fixed speed again (about 30 degrees); then switching to gps tracking again
3. Somehow combining 1 and 2 via some decision making process to imitate what a real sailor would do (try to keep close to the course without sacrificing too much speed).

I attached a picture from our sailing simulator which shows some of the boat parameters we used to control a tack (beta is the apparent wind angle). We didn't have a controller though, were using prescribed inputs instead.

[image]

smaria, 2016-02-04:
The tack will be executed by a dedicated node ( #12 ). While the sailing state is 'tack_to_XXXX', messages from this node overwrite the sail and ruder PID. The goal_heading message is ignored during this phase, but becomes active as soon as the tack is over.

The goal of this ticket is to work out how the transition between normal sailing and tacking is done - how do we ensure a smooth transition from the tack to the normal heading control.

- How is it determined that a tack is finished (by the heading_control node? By the tack node?)
- What steps are needed to transition back smoothly? 
- should the sheet be let out slowly before a tack is finished? This could use the messages that the PID control continues to send, and approach them within a few steps

Based on the notes above I would say the basic implementation should be:
The goal heading after the tack is determined in heading_control before the tack state is entered.
The tack is finished, when the goal heading after the tack is reached within an error margin.
The first published goal heading is the goal heading determined before the tack.

The tack node itself is not involved in determining if a tack is finished, 

The next steps:
- Timeout of the tack (so we can't get stuck) #26
- Deal with wind changes #27

takluyver, 2016-02-17:
In the heading control code that I've written, it keeps the sailing state as 'tack_to_*_tack' until we reach the minimum angle we can sail towards the wind (which I'm calling the 'beating angle' in code). Throughout this time, the goal heading is set to that angle. Once we reach that angle, the state returns to 'normal', and the heading updates to drive us towards the waypoint (unless the waypoint is too close to the wind, in which case the heading stays at the beating angle).

closed 2016-02-17 by takluyver