Maritime-Robotics-Student-Society/sailing-robot #22: Write wind_estimate node

smaria, 2016-02-03:
the wind direction needs to be estimated from the magnetometer under the wind vane.
Can further information about wind-speed be read in combination with other sensors?

UnnamedMoose, 2016-02-04:
The wind vane will read apparent wind speed, while for navigation/optimal course estimation true wind speed will be more useful; hence this wind_estimate will also need to know the boat heading and speed.

smaria, 2016-06-29:
There are a number of wind directions defined, and split into tickets:
- Average the sensor values directly, (topic [wind_direction_apparent](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-conventions#wind_direction_apparent)) : #70 
- Get a longer term average wind direction, (topic [wind_direction_average](https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-conventions#wind_direction_average)) # 74

Left for later, when a speed and wind speed estimate is available: wind_direction_real
(at this point the wind_direction_average should be switched to using the real wind speed)

smaria, 2016-08-18:
we now average our wind speed over two time periods;
there will be no wind speed sensor on the boat, but we might have speed information from land. This is for a later time though.

closed 2018-09-01 by Nanoseb