Maritime-Robotics-Student-Society/sailing-robot #28: New website under University domain

tsaoyu, 2016-02-04:
It is necessary to setup a public website for our team, ideally a WordPress site. And university provide some free host service via `blog.soton.ac.uk` and I am planning to apply a subdomain like `www.soton.ac.uk/roboticsailing` or `www.soton.ac.uk/robotsail`. We'll need find an academic mentor (person who can authorise the application) and fill out the [form].(https://sotonproduction.service-now.com/soton/it_rq_new_website )

bigboateng, 2016-02-05:
Yes I think a wordpress site is a great idea as opposed to creating one from scratch. Saves time and wordpress has a nice interface for non-developers to add and edit content, which is what the Management team wants to do. I (and others who are interested) can setup the initial pages with content like logo's, contact details, about the team, etc then the management team can take over and content as time moves on.

smaria, 2016-02-05:
just a note: github can also be used to set up a page, a friend of mine set up the fencing website like this: http://www.sotonfencing.co.uk/ 
https://github.com/sotonfencing/sotonfencing.github.io

tsaoyu, 2016-03-02:
I have set up an wordpress site for our team at 
http://sailrobot.wordpress.com
 
Simple because 

- WordPress is more user friendly options
- University domain is only open to research groups

> Comments:
> 
> 26-02-2016 11:32:21 GMT - Kate WoodPublic notes
> 
> Hi,
> 
> Elisavet Hadjiodysseos submitted a request yesterday for a website for this team and I advised her to contact SUSU as it's a student group rather than a University research group.
> 
> Does this request relate to the same request she put in yesterday? If so, is it okay if I close this off and you talk to her about what she's doing?
> 
> Thanks
> 
> Kate
> 
> Regards,
> 
> iSolutions

I have already set Elisavet as administrator and Pier as editor as our website. I think they could take over this website soon.

closed 2016-03-02 by tsaoyu