Maritime-Robotics-Student-Society/sailing-robot #31: Handle running before the wind

takluyver, 2016-02-17:
The current code will aim the boat directly towards the waypoint unless that's too close to the wind for direct sailing. But if the wind shifts slightly across the heading to the waypoint - e.g. if we're heading due south, and the wind direction changes from 179 to 181, the code will currently decide that the angle change means we should switch tack, which could lead to a very wiggly course.

When our desired heading is close to 180 degrees from the wind, we need some special handling that allows us to go some way beyond 180 degrees on the same tack, to avoid unnecessary tacking. But we want to avoid unexpected jibing. Key factors are:

- The wind angle which will cause us to jibe
- Estimated variability in wind direction and our estimate of true wind direction

Nanoseb, 2016-09-03:
this is not done, but is not really needed, at least for WRSC2016 (because we can already jibe)

takluyver, 2016-09-03:
Actually, you sort of did this in e091a5b4e9ba3dcfe4feef3eaa01f9049a6586ab. The key point is that it won't try to tack if it has to turn from +170 to -170 degrees relative to the wind.

When I wrote the issue I thought we wanted to avoid jibing, but I now think that for our boat jibing doesn't really matter. So let's close this; we can create a more specific issue if there's still something we want to do.

closed 2016-09-03 by takluyver