Maritime-Robotics-Student-Society/sailing-robot #33: Finishing a tack

takluyver, 2016-03-02:
The current code, when it detects the tack is complete, immediately sets the sailing state to 'normal'. If we're tacking to change direction (rather than tacking upwind), it will therefore keep straight on turning to the desired heading. Sailing experts: is that a good idea, or do we want to put the rudder straight and accelerate for a bit before continuing the turn?

pierbiag, 2016-03-03:
I don't understand what you mean by "tacking to change direction". A tack strictly means changing the side the wind is coming from relative to the boat by going directly upwind and keeping turning (crossing the dead zone). If you mean tacking and than keep turning further than a normal tack, it's better to do the maneuver altogether: the most the relative wind angle approaches 90°, the faster the boat accelerates and goes (meaning there is no point in keeping the heading near the dead zone if not needed).

takluyver, 2016-03-03:
Thanks - I did mean tack and keep turning. So there's nothing to be fixed here.

closed 2016-03-03 by takluyver