Maritime-Robotics-Student-Society/sailing-robot #35: wind_apparent parameter as a tuple

pierbiag, 2016-03-03:
Please consider changing the format of the  wind_apparent parameter to a 2-elements tuple: wind angle and wind speed.
This because at some point we will likely have a wind speed sensor, and because I don't know how hard it will be to change it later, I thought it was a good idea to mention it now.

smaria, 2016-03-03:
Good point, though it is a topic, not a parameter, see:
https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Naming-Conventions#wind_apparent

After the first dry land test I will discuss with the firmware and hardware team if we want this in separate messages or if this information will come from the same node.

smaria, 2016-03-05:
I decided against tuples since we don't know if wind speed and wind direction will be handled by the same node. In the documentation we now have apparent, real and average wind direction and apparent and real wind speed.

closed 2016-03-05 by smaria