Maritime-Robotics-Student-Society/sailing-robot #36: Make raspberryPi interface that can be accessed over wifi

smaria, 2016-03-03:
It would be fancy if we could use a website (or maybe even something via ROS?) to control the sailing robot. That way we can control it via a tablet or from our own laptop.

bigboateng, 2016-03-03:
We can use firebase (https://www.firebase.com/) . It allows real time data streaming across platforms. So we can do real time on and android app as well as website. I've used it before in past projects. I can work on it once we high priority stuff out the way. Just need the python library for ros, javascript library for website and java library for app.

pierbiag, 2016-03-04:
Ideally we want something that doesn't have to connect to an external server (both for lag and foreseeable connection problems at sea). I think a simple html page hosted on the raspi (ideally configured to act as an access point) will be a good option.

bigboateng, 2016-03-04:
I guess so, but Firebase is really fast though. 
 if you host web page on pi, then on the sea how would you connect to it if its wifi/the wifi its connected to is not in range?

pierbiag, 2016-03-04:
The normal commands are given through the low frequency radio modules, but when within range, the raspy itself becomes an access point for the other devices. (No need for Internet)

bigboateng, 2016-03-04:
I get you

NanaOKA, 2016-03-14:
Established connection via ssh using an access point on the Raspberry Pi. This will allow us to possible control the robot wirelessly using a ROS network consisting of the Raspberry Pi and a laptop

tsaoyu, 2016-06-04:
We are currently using AP mode for remote control of our sailing robot.

closed 2016-06-04 by tsaoyu