Maritime-Robotics-Student-Society/sailing-robot #42: merge all packages together

smaria, 2016-04-27:
there is no need to have the packages split up

smaria, 2016-04-27:
Thinking about this one - maybe it is better not to do it, but split into two packets: 

- boaty_mcboatface
- sailing_robot

hardware dependent nodes are in boaty_mcboatface
hardware independent nodes are in sailing_robot

everything from sailing_robot can be used for other sailing robots as well
everything in boaty_mcboatface might be better of rewritten if we get a new boat (and we may want to keep boaty_mcboatface so we can still use the original boat)

takluyver, 2016-04-27:
On the other hand, if a new boat used logically similar components, we may be able to reuse almost all the code with some changes to parameters. It's hard to know how reusable stuff is until we have two boats to run.

smaria, 2016-05-17:
Ok, lets go with one package for now and split boats away if needed

smaria, 2016-05-17:
merge was done and tested, please review, merge

tsaoyu, 2016-05-17:
Correct the name of package from  `src/sailing_robot/src/high_level -> src/sailing_robot/src/sailing_robot`

takluyver, 2016-05-18:
Closed by #59

closed 2016-05-18 by takluyver