Maritime-Robotics-Student-Society/sailing-robot #50: Travis tests fail

takluyver, 2016-05-01:
I can't figure this out. The tests are failing with:

```
======================================================================
ERROR: Failure: ImportError (No module named highlevel_control.heading_planning)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/loader.py", line 411, in loadTestsFromName
    addr.filename, addr.module)
  File "/usr/lib/python2.7/dist-packages/nose/importer.py", line 47, in importFromPath
    return self.importFromDir(dir_path, fqname)
  File "/usr/lib/python2.7/dist-packages/nose/importer.py", line 94, in importFromDir
    mod = load_module(part_fqname, fh, filename, desc)
  File "/home/travis/catkin_ws/src/sailing-robot/src/highlevel_control/tests/test_heading_planning.py", line 5, in <module>
    from highlevel_control.heading_planning import HeadingPlan, angleAbsDistance
ImportError: No module named highlevel_control.heading_planning
```

I can't reproduce this on the ROS Ubuntu VM. The first commit where it fails is a64cc2d741f6adfb28cb5eda0e4f96097fc86f49, but that doesn't appear to change anything related to this, so maybe it's something else, like a ROS update that happened to appear at that time.

I'm hoping someone with more ROS knowledge than me can figure it out.

takluyver, 2016-05-01:
Looks like it is a change in an external update of some kind. I made a new branch so that Travis would re-run the test for the last commit that [passed before](https://travis-ci.org/Maritime-Robotics-Student-Society/sailing-robot/builds/124574095), and now the tests [fail on that]https://travis-ci.org/Maritime-Robotics-Student-Society/sailing-robot/builds/127047883] as well.

Looking at the build logs, it looks like:

- `python-rospkg` 1.0.38 -> 1.0.39
- `python-wstool` 0.1.12 -> 0.1.13
- `ros-indigo-catkin` 0.6.16 -> 0.6.18

tsaoyu, 2016-05-11:
The travis build is originated from https://github.com/felixduvallet/ros-travis-integration 

It explained the mechanism inside the travis test. Check it if necessary.

tsaoyu, 2016-05-11:
Finally fixed the problem by source the setup file in our catkin work space. Like what have been mentioned in felixduvallet/ros-travis-integration#5

closed 2016-05-11 by tsaoyu