Maritime-Robotics-Student-Society/sailing-robot #51: Implement sailing area limit

smaria, 2016-05-04:
We need to stop the vessel from going into too shallow water by accident, or leave a defined box, so we need some supervising solution that will overwrite the current waypoint??

How do we best implement this?

takluyver, 2016-05-04:
A simple solution would be to define a box, and if we go outside it,
override normal directions and sail back towards the centre of the box.

Complexities: what if the centre of the box is upwind? What if we get back
in and the normal directions immediately turn us round to leave again?

smaria, 2016-05-04:
I have doubts this will work with #57

takluyver, 2016-09-03:
I've made it so that we can specify a safety zone (make `navigation/safety_zone_ll` a list of lat/lon pairs), and if it detects it's near the edge, it will head for the centroid of the safety zone. This should work in most scenarios, but there are cases where it could fail:

1. If the centroid is outside the safety zone (e.g. if the safety zone is L shaped)
2. If we end up in a downwind corner of the safety zone, we may have no path back without leaving the zone.

I've tested that this doesn't break anything when we don't specify a safety zone; I haven't yet tested specifying a safety zone.

closed 2016-09-03 by takluyver