Maritime-Robotics-Student-Society/sailing-robot #61: Add wifi setup to launch script on RPi

smaria, 2016-06-04:
This way we don't have to be connected to the Raspi via wired connection after reboot. This involves:
- updating the launch script
- updating the image available

smaria, 2016-06-04:
instead of updating the image we should update the image setup instruction

tsaoyu, 2016-06-04:
Currently, the WiFi connection is depends on the model of our WiFi dongle. I couldn't figure out a uniform way to update method for potential different adapters. 
But, WiFi AP setting would be documented in the online WiKi.

Nanoseb, 2016-06-04:
we can write a script to detect le dongle (by parsing the output of `lsusb` it is probably quite straight forward)

tsaoyu, 2016-06-04:
The problem is related to the drivers issue. 
In order to use a specific type of dongle, we need use different version of `hostapd`.
With different model of dongle, we need to remove and recompile a specific version of `hostapd` which makes thing complicated. 
I am willing to try if it possible to use `lsusb` to automatic detect and setup of AP.

tsaoyu, 2016-06-04:
https://github.com/theOtherLuke/rpi-access-point/blob/master/jenssegers_rtl8188_hostapd_install.sh

Nanoseb, 2016-06-04:
Ok, so instead of compiling every times, you can just uninstall the previous one and run make install. The run at boot part of this script will not work since we are using Jessie (systemd).

tsaoyu, 2016-06-04:
My idea was shrink the image we used today as the "base" image and forget about the WiFi setting. Even we use `lsusb` to identify the model we are using, we couldn't quickly find the corresponding custom-made `hostapd` and drivers. This means, we will always keep using a single model of WiFi dongle until there is huge modification on our current way of wireless communication.
So I would like to shrink our image a little bit and use today's image as the base image as the back up.

NanaOKA, 2016-06-06:
Added bash script fro setting up wireless access point for this dongle http://www.modmypi.com/raspberry-pi/accessories/wifi-dongles/wifi-dongle-nano-usb

tsaoyu, 2016-06-06:
Drivers on current RPi is `rt1871xdrv`, I think this script can only make that type of dongle works.

`pi@raspberrypi:~/sailing-robot $ vi /etc/hostapd/hostapd.conf `

```
# Basic configuration

interface=wlan0
ssid=Sail_robot
channel=1
#bridge=br0

# WPA and WPA2 configuration

macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=3
wpa_passphrase=sailrobot
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

# Hardware configuration

driver=rtl871xdrv
ieee80211n=1
hw_mode=g
device_name=RTL8192CU
manufacturer=Realtek

```

NanaOKA, 2016-06-06:
Makes sense to use only one type of dongle as I don't think any spares we get will be off a different type.

tsaoyu, 2016-06-06:
The thing is I used to think we couldn't make our dongle (Edimax) works until I try that with the custom made drivers. So let's stick with our plan, and use the dongle you posted here as the only type of WiFi adapter for our project. 
And I will use @Nanoseb 's 8GB image as our base image for the backup purpose.

That image will be named with date and Raspbian version. All the setup script will be moved to `utilities` folder. Once we recovery the image, we should also run all the scripts created(modified) later than the image size. 

In this way, we don't need to worry to much about our base image. And I'll upload that image to the `sailrobot` google drive.

closed 2016-06-06 by tsaoyu