Maritime-Robotics-Student-Society/sailing-robot #63: Work out what's wrong with heading/tacking calculation

takluyver, 2016-06-05:
Yesterday's test ended with us going round and round in circles a lot. This might be great for the virtual anchoring task, but we were trying to sail to a waypoint. Something seems to be up with the heading or tacking calculations.

smaria, 2016-06-23:
This was investigated further at the last dry land test, as far as I recall we narrowed it down to the wind direction needing averaging ( #70 ) and the compass having an offset/sign error (# ???)

takluyver, 2016-06-23:
Sign or offset issue is/was #71.

takluyver, 2016-06-27:
I've rewritten the heading control (which decides when to tack) to hopefully address the overenthusiastic tacking we were still seeing.

The new logic calculates at each time step if the other tack would take us closer to the waypoint. But it doesn't then use that directly, but instead adds it to a running poll. It tacks when the poll reaches 75% of samples wanting to tack over the last 10 seconds (these values can be made configurable if this mechanism works). This should mean:

- No tack for at least 7.5 seconds after finishing a tack - but unlikely my quick and dirty solution on Saturday, the rudder PID will get a chance to steer the boat.
- Better coping with fluctuating wind by preferring the tack we're currently on.

For now, I've simplified it by cutting out any calculation of lay lines, relying on the delay this provides to keep us from constantly tacking. I suspect we'll probably need to add a bit more complexity, but that was in need of a rewrite from our 'tack lines' idea anyway (#58), so I'll work out how best to do it once we've seen how the boat behaves with this code.

Nanoseb, 2016-06-27:
This seems a really good idea. nice!

takluyver, 2016-07-20:
Navigation was basically successful on Sunday, though there's still some optimisation to do.

closed 2016-07-20 by takluyver