Maritime-Robotics-Student-Society/sailing-robot #65: Better way to monitor robot

takluyver, 2016-06-05:
It's not currently easy for us to see basic things like where the robot thinks it is in relation to its waypoint. SSHing in and running `rostopic echo` only lets us monitor one topic per terminal, but running ROS' rqt interface requires a full ROS desktop build. Additionally, lat/long values don't intuitively mean much to us for navigating on a small scale - we need distances in metres.

tsaoyu, 2016-06-05:
It is possible to running ROS on multiple machines, and I assume we can run `rqt` on the laptop node. 

We also need to create a new topic `/position_error` for distance error.

smaria, 2016-06-05:
Running ROS on multiple machines is not implemented so it is easy (yet - it is a matter of setting some variables more permanently, -> #66 )

tsaoyu, 2016-06-14:
https://github.com/nasa/openmct

takluyver, 2016-06-25:
The HTML dashboard is a good start on this. There are still improvements we can make to get info more intuitively (e.g. I want to have a compass-style view showing heading, wind direction, waypoint direction and goal heading), but I'll close this and open separate tickets for such things.

closed 2016-06-25 by takluyver