Maritime-Robotics-Student-Society/sailing-robot #71: Investigate the sign error/offset

tsaoyu, 2016-06-15:
[]Investigate the sign error on rudder command and wind direction
[]Investigate the offset issue on wind direction

Nanoseb, 2016-06-15:
For the offset value on the wind direction, I think it is better to handle it in the calibration script because depending on magnets directions it will add/subtract 180 degree. I'll add that.

tsaoyu, 2016-06-16:
Now the sign issue have been resolved. 
[image]
Need hardware test to figure out an suitable PID value.

Nanoseb, 2016-06-16:
If everything is in degree and we are using only P, I think a P of 1 is a good start. The final value will probably be lower than that.

smaria, 2016-06-22:
please include all things decided/noted here in the wiki, since that is where we will check later!

tsaoyu, 2016-06-22:
Anti-windup have been implemented in `pid_rudder` node.

closed 2016-06-22 by tsaoyu