Maritime-Robotics-Student-Society/sailing-robot #75: Investigate the discontinuity of power bank supply

tsaoyu, 2016-06-22:
We find that our power bank stop to supply power when the current is low. This will cause servos lose power after a while. Solution can be:

* Use AA batteries
* Maintain the lower power supply level with resistor

This problem should be fixed before this Saturday (25/06).

smaria, 2016-06-23:
Can you describe that in a bit more detail? The power bank still supplies the Raspberry Pi, but the servos don't get powered?

tsaoyu, 2016-06-23:
We use two power bank to power RPi and servos separately. 
Once the current supply from power bank is low, the control circuit inside power bank will cut off power supply. 
When servos idle, they draw so little energy from power bank that switch off the power bank.

smaria, 2016-06-23:
It should be fine to run the servos and the RPi off the same supply - originally there was some concern that the servos might draw much higher currents than the RPi, leading to sudden voltage drops, but it seems to me the maximum output current of the power supply is large enough. Based on the last lake test, the power consumption of the servos is insignificant compared to the continuous current draw of the RPi.

smaria, 2016-06-24:
Ok, today taught us better: Running the servos can make the RPi reboot. For now we are using AA batteries.

tsaoyu, 2016-06-28:
Conclusion is power supply separation is necessary. And I am going to order another 16 AA batteries.

closed 2016-06-28 by tsaoyu