Maritime-Robotics-Student-Society/sailing-robot #77: Permission Issue on RPi native servo control node

tsaoyu, 2016-06-23:
@Nanoseb @pierbiag and I made a simple servo control node use native PWM control on RPi.
This hardware PWM control works pretty well even when computational load is high on the pi.
The only draw back of my current version of driver requires `sudo` right, and it mess up with `rospy` because the `PATH` of `root` is different from our current user `pi`.

Before I figure out this permission issue, we'll still use Arduino for servo control.

Nanoseb, 2016-06-23:
https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root/46235#46235 I think it is the cleanest way we can achieve that.

TL;DR give read/write access to the group gpio to /dev/mem to control gpio pins without root access

tsaoyu, 2016-06-23:
Another issue the bad control quality using `wiringPi`
Suppose we are using 50 Hz PWM to control the servo, useful duty cycle is between 5-10% percent (1ms~2ms correspondingly). And the `wiringPi.pwmWrite(int dutycycle)` is 0-1024, so we only have 5% of them (50 steps) for 180 degrees range. That is quite bad performance according to my test. 
When I increase the PWM frequency to 100 Hz, Futaba servo stop responding. I am not sure if this servo only support 50 Hz. 
I will test the rudder servo on the laser to see if it possible to improve the performance.

For the permission issue, I think it is similar to serial problem
```
pi@raspberrypi:~/Desktop $ groups pi | grep gpio
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio
```

I will look at this issue later. 
Cheers

Nanoseb, 2016-06-24:
the rpio library checks if it is run as root but not if actually /dev/mem has the right permissions. I think we can fork it to skip this check and it will probably work fine

Nanoseb, 2016-06-24:
we can try with this: https://github.com/Nanoseb/RPIO

tsaoyu, 2016-06-28:
This piece of code suppressed the notification of permission issue. But servo stop responding because `/dev/mem` still not accessible by the code.

Nanoseb, 2016-06-28:
but have you changed the permissions on this device ? (it might be needed at every boot )`sudo chmod 666 /dev/mem` (or you can add the proper user to the group and do a 660)

tsaoyu, 2016-06-28:
Yes, I have done that once. But I think seems not a good idea to open our physical memory  permission `/dev/mem` to users.

Nanoseb, 2016-06-28:
you can achive the same thing by just adding the gpio user as an owner of the device (as explain in the stackexchange post)

tsaoyu, 2016-06-28:
Will check that tonight.

BTW backup RPi is running and accessible on `10.15.36.116 `(intarnet).

tsaoyu, 2016-06-28:
This is my attempt:
```
pi@raspberrypi:~/pigpio-master $ sudo chown root.gpio /dev/mem && sudo chmod 666 /dev/mem
pi@raspberrypi:~/pigpio-master $ rpio
Traceback (most recent call last):
  File "/usr/local/bin/rpio", line 4, in <module>
    __import__('pkg_resources').run_script('RPIO==0.10.1', 'rpio')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio", line 353, in <module>
    
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio", line 147, in main
    
  File "build/bdist.linux-armv7l/egg/RPIO/__init__.py", line 115, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 7, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 6, in __bootstrap__
RuntimeError: No access to /dev/mem. Try running as root!
pi@raspberrypi:~/pigpio-master $ ls -al /dev/mem
crw-rw-rw- 1 root gpio 1, 1 Jun 28 20:45 /dev/mem
pi@raspberrypi:~/pigpio-master $ groups pi | grep gpio
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio

```
I do think expose `/dev/mem` is not a clean solution. 

The next library I want to try is `pigpio` http://abyz.co.uk/rpi/pigpio/index.html

Nanoseb, 2016-06-28:
I think the problem will be the same with every library, to set gpio some rights are needed no mater what is the implementation.

takluyver, 2016-06-28:
Other people must have run into this, surely? Is there any info on what people do to use gpio?

tsaoyu, 2016-06-28:
Yes, I agree with you that GPIO operation do need right permission. But it is also depends on the implementation of the library. I can understand and give permission to `gpio` memory like this
```
pi@raspberrypi:~ $ ls -al /dev/gpiomem
crw-rw---- 1 root gpio 244, 0 Jun 28 22:47 /dev/gpiomem
```
But fully opened permission `/dev/mem` will cause some unexpected issue. 

For now, I have test the `pigpio` library. The implementation is much more clear and securer than `RPIO`.
Basically the idea is let `pigpiod` running as daemon on `localhost(8888)` with `sudo` rights. And every GPIO command (e.g. hardware PWM) go through this service.

takluyver, 2016-06-28:
If the best answer is that something needs to be run as root, then I agree that that should be a simple service which acts on behalf of the programs that need to do things.

tsaoyu, 2016-06-28:
For Thomas' question: Other people do run into this problem, but occasionally, they also using ROS at the same time. They don't have `PATH` issue when blindly running scripts as `root`. 

GPIO memory can be accessed by `gpio` groups. So usually software GPIO wouldn't is fine without `sudo`. But in our case we need high quality hardware PWM wave to control our servos, those conventional library is no longer able to meet our requirements.

tsaoyu, 2016-06-29:
This problem is solved with `pigpio` library and new `servo_control` node.

closed 2016-06-29 by tsaoyu