Maritime-Robotics-Student-Society/sailing-robot #81: test that we can use RPi without internet

smaria, 2016-06-25:
we should test what changes need to be made if we switch from using git on the RPi to copying over code from our machines in a separate folder

takluyver, 2016-06-25:
It is possible, with a bit of config, to push code with git directly from our computers, without needing an internet connection. This might be preferable if it involves changing less than pushing code to a separate folder.

takluyver, 2016-06-25:
Places where we've hardcoded a path to the `~/sailing-robot` folder:

- Sourcing setup.bash in `~/.bashrc`
- In `dashboard` node for static file `index.html`
- In at least one launch file, for the directory where `rosbag` is to record.

Suggestion: rather than rewriting the paths in several different locations, we should make `~/sailing-robot` a symlink that can we can switch between `sailing-robot-git` and `sailing-robot-nogit`.

takluyver, 2016-07-03:
I have git configured so I can push directly to the raspi. The config in `.git/config` looks like this:

```
[remote "raspi1"]
    url = pi@sailbot:sailing-robot-bare
    fetch = +refs/heads/*:refs/remotes/raspi1/*
```

* On the working computer, push with `git push raspi1`.
* On the raspi, then `git pull bareclone`

(This requires two steps, because pushing to a normal git repo with a working directory fails if it would update the current branch. Therefore we have a bare clone on the raspi at `~/sailing-robot-bare`, which we push to and then pull from)

takluyver, 2016-07-18:
I've documented the steps to set up git for this:

https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Git-push-directly-to-the-raspi

closed 2016-07-18 by takluyver