Maritime-Robotics-Student-Society/sailing-robot #85: Ensure there is enough free space on SD card for rosbag

takluyver, 2016-06-26:
We were getting messages yesterday that rosbag wasn't recording because it detected that there was <1G of free space left. We need to ensure that the recording is actually able to record, and that we don't run out of space.

If we're using SD cards larger than the image, one possibility is to make another partition on the card in the remaining space, and tell rosbag to write files there.

Nanoseb, 2016-06-26:
The SD card is 32G right now, we can even expand the file system instead of making a new partition.

takluyver, 2016-06-26:
My idea to make a separate partition was in case we need to copy the image from the raspi again - though I think we should aim to avoid copying lots of images, it's a useful possibility to have. If it's a second partition, I assume we can just copy the first partition off.

Nanoseb, 2016-06-26:
We have found a way to shrink an image to its 'useful' content, and we already have a 7G image (on the google drive), so we can expand this one and if we want to make a other sd card we just have to get the 7G one and run the scripts from the utility folder to have it at the same state as the one we already have.
But yes a new partition is also not hard to handle (just make sure it is mounted on boot.

takluyver, 2016-06-26:
That's good to know. Presumably it still takes longer to make a 32GB image from the card than it does an 8GB image, though.

tsaoyu, 2016-06-28:
We can expand the file system. There is 8 GB card with last copied image and 32 GB empty card for back up. I'll update that 8 GB card every Wednesday.

tsaoyu, 2016-07-01:
File system now expanded.

```
pi@raspberrypi:~/sailing-robot $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  6.0G   22G  22% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           463M     0  463M   0% /dev/shm
tmpfs           463M  6.3M  457M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           463M     0  463M   0% /sys/fs/cgroup
/dev/mmcblk0p1   63M   21M   43M  33% /boot
tmpfs            93M     0   93M   0% /run/user/1000
```

closed 2016-07-01 by tsaoyu