Maritime-Robotics-Student-Society/sailing-robot #86: Work out why GPS is unstable

takluyver, 2016-06-26:
Our GPS position was fluctuating by tens of metres yesterday. We need a fix accurate to within a few metres.

We thought that putting an antenna outside the boat casing might help.

Nanoseb, 2016-06-28:
Because we have an imu, we can use a kalman filter (https://en.wikipedia.org/wiki/Kalman_filter) to have more accuracy on the gps. The math behind it is probably quite complex, but there is probably some nice python libraries to do that

Nanoseb, 2016-06-28:
do we have any indication on the signal quality of the gps? (number of satellite for example), with that we can see where is the best place for it.

tsaoyu, 2016-06-28:
ROS package `robot localization` can do this job nicely. One thing we need for the Kalman filter is an approximate linear model of our boat. 
http://wiki.ros.org/robot_localization 

I am not sure if we are currently recording the number of GPS satellite or not. But I suppose it would be hard to get that data using `cgps`.

takluyver, 2016-06-28:
The data we get in the GPS node does include the number of satellites it's tracking, but we don't currently pass it on through ROS (and annoyingly, there doesn't appear to be a field for it in the NavSatFix message).

However, let's test it with a GPS antenna outside the boat before we try combining data from multiple sensors. The extra complexity will give us more things to debug, and if the crucial input is dodgy, the output probably won't be that great.

tsaoyu, 2016-06-28:
Yes, we may go for "physical" solution first and then try smart EKF (or UKF) approaches. 

We also have ultimate backup plan https://emlid.com/shop/reach-rtk-kit/

tsaoyu, 2016-07-06:
So sad R.E. Kalman passed away several days ago. http://hungarytoday.hu/news/renowned-hungarian-scientis-rudolf-kalman-dies-aged-86-46732
R.I.P.

smaria, 2016-08-18:
GPS antenna is now outside of boat, waterproofed. At least for now this is a good enough solution.

smaria, 2016-09-24:
Don't think this is a problem these days.

closed 2016-09-24 by smaria