Maritime-Robotics-Student-Society/sailing-robot #89: Work out GPS parsing issues

takluyver, 2016-06-30:
Using the GPS yesterday, I saw:

- pynmea2.ParseError
- pynmea2.ChecksumError
- UnicodeDecodeError (this one is probably a bug in pynmea2)

At a minimum, we need to catch errors so they don't take down the GPS node.

takluyver, 2016-07-02:
I have (hopefully) ensured that these kinds of errors will be caught and the GPS node will continue publishing. I'm logging details through ros so that if errors reappear, I'll have more information to help make a proper fix.

Of course, I can't readily test this without the GPS receiver plugged in.

takluyver, 2016-07-03:
Error details logged today for UnicodeDecodeError:

```
[ERROR] [WallTime: 1467549410.551766] Error parsing GPS data.
buffer=u''
line='5)\xff$GNRMC,,V,,,,,,,,,,N*4D\r\n'
Traceback (most recent call last):
  File "/home/pi/sailing-robot/src/sailing_robot/scripts/gps", line 37, in pos_publisher
    batch = gps_reader.next(line)
  File "/usr/local/lib/python2.7/dist-packages/pynmea2/stream.py", line 30, in next
    lines = (self.buffer + data).split('\n')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 2: ordinal not in range(128)
```

smaria, 2016-08-22:
We can also make the GPS node a node that respawns automatically; this will give a little more delay but should work long before the GPS looses fix.

http://wiki.ros.org/roslaunch/XML/node respawn = true

Are we waiting for something to close this?

takluyver, 2016-08-23:
I was leaving it open to try to figure out what was causing the errors we see, but my code to catch errors and carry on seems to be working well enough, so this isn't high priority.

closed 2016-08-23 by takluyver