Maritime-Robotics-Student-Society/sailing-robot #91: Wi-fi signal issues

takluyver, 2016-07-04:
We were getting very spotty wifi connections to the boat during yesterday's lake test.

We think this might have been because the wifi antenna in the boat was too low down; we should investigate putting it on top of the boat.

smaria, 2016-07-06:
Let's for now keep using the current adapter (since there is limited compatibility with the RPi), and use this to put it on top of the mast: 
https://www.amazon.co.uk/AmazonBasics--Male--Female-Extension-Cable/dp/B00NH136GE
@NanaOKA do you know of another wifi adapter that might have a better range at the top of your head?

tsaoyu, 2016-07-06:
I am not sure this is related to WiFi power management system on RPi.
We can try http://www.modmypi.com/blog/disable-wifi-power-management

tsaoyu, 2016-07-06:
If power isn't the issue, I would try http://www.modmypi.com/raspberry-pi/accessories/wifi-dongles/wifi-dongle-ultra-long-range-high-gain-w-5dbi-antenna/?gclid=CjwKEAjw2PK7BRDPz5nDh9GjoGcSJAAybcS3dqAlzwF121kNSGkBLr_oZQMbRAskxDbuWYJNy5OvCRoC78_w_wcB

smaria, 2016-07-06:
My impression was that it is mainly an issue with the antenna being underwater. The current antenna would definitely be lighter on the mast. Then again, maybe worth buying in case the small antenna doesn't work even on top of the mast.

tsaoyu, 2016-07-13:
Our RPi is now equipped with high gain antenna and USB extension cable.

closed 2016-07-13 by tsaoyu