Maritime-Robotics-Student-Society/sailing-robot #93: rename nodes consistently

smaria, 2016-07-07:
our node naming needs some consistency!
We will start using prefixes:

- debugging_
- dummy_
- simulation_
- sensor_ 
- actuator_

Nanoseb, 2016-07-07:
I would say:
```
blink_on_sailing_state -> debugging_blink_sailing_state
dashboard -> debugging_dashboard
gps -> sensor_gps
gps_log -> debugging_gps_log
imu-calibrated -> sensor_imu
pid_rudder -> actuator_rudder
pid_sail -> actuator_sail
publish_apparent_wind_direction -> dummy_wind_direction_apparent
publish_heading -> dummy_heading
publish_position -> dummy_position
servo_control -> actuator_servo_control
tack -> hilevel_tack
tasks -> hilevel_tasks
wind_direction_average -> hilevel_wind_direction_average
wind_direction_calibrated -> sensor_wind_direction
```
I added hilevel to the prefixes.

smaria, 2016-07-07:
we have a couple of 'archive' launch files, I am currently not planning to migrate them to the new names.

smaria, 2016-07-07:
also, do we want to change the names of the nodes that are launched in the scripts as well? they are often a bit different from the names of the scripts.

smaria, 2016-07-07:
> tack -> hilevel_tack
> tasks -> hilevel_tasks
> wind_direction_average -> hilevel_wind_direction_average
> 
> I added hilevel to the prefixes.

I don't really see much connecting those three nodes, besides them all not being in any of the other categories, so for now I would not put them together (but maybe choose a longer, more descriptive name for them)

Nanoseb, 2016-07-08:
I think it is better to have node names = script names, I don't see any reason not to do that. 

For the `hilevel` prefix, maybe we can change that to `other`, and use a more descriptive name. But having a prefix for all of them is probably a good thing.

smaria, 2016-07-08:
I can agree on putting tack and tasks in one bucket, but the wind sensor analysis imho doesn't fit. If it has to be grouped I'd put it in with sensors.

I also added sub-groups for the actuators: actuator_demand and actuator_driver

smaria, 2016-07-10:
> they publish values which the real actuator node controls. I would reserve the name actuator_ for the node(s) that interface with the actuators, and call these...I don't know, 'processing_'?

> Likewise, I think we should reserve sensor_ for nodes that read data directly from sensors. I'd call this one something like processing_smoothed_wind_direction.

Trying to understand this as an alternative to the current naming: This proposes to split into actuators, sensors, processing, debugging, dummy, and simulation.

issues I have with this:
- I find 'processing' not descriptive
- I don't see a clear connection between smoothing the wind direction and generating an actuator demand, they seem more thrown together because they are left

takluyver, 2016-07-10:
I agree that 'processing' is not very descriptive, and it is certainly something of a 'misc' category. What they have in common is that they subscribe to ROS topics, process data, and publish to other ROS topics. This overlaps with 'hilevel', and in a larger system, presumably there would be many nodes like this.

I don't have a way I'm really happy with to categorise these. But among the nodes that run when we're actually controlling a boat, the only clear distinction I see is between nodes that talk to hardware and nodes that don't. That's meaningful because nodes that don't talk to hardware can be tested in a simulation. So I'd rather preserve that distinction, even at the cost of creating an 'everything else' category, rather than trying to push the leftovers into 'sensors' and 'actuators'.

smaria, 2016-07-10:
The flexibility between hardware and simulation is a good point. Would it be satisfactory for you to have sensor_driver and actuator_driver as prefixes for the parts that are replaced? 
I am quite attached to splitting the lowlevel analysis and control parts into sensor and actuator related bits, since I find it makes the structure of the firmware understandable.

Otherwise I would probably not even call them sensor or actuator, maybe 'actor', 'physical', or 'hardware', and then split the rest in 'analysis', 'control' and 'plan'

takluyver, 2016-07-10:
Go with what makes sense to you; so long as it doesn't keep changing, I'll work out what the names mean and work with it.

smaria, 2016-07-10:
I am going with the sensor/actuator naming scheme. Reason: sensor_imu doesn't actually publish imu data, it publishes already processed heading data, making it difficult to classify under the other scheme. Whilst this may need splitting some point, I am not going to split it before implementing the re-naming.

I will name it sensor_driver_imu to mark it as a node that contains a driver.

closed 2016-07-13 by Nanoseb