Maritime-Robotics-Student-Society/sailing-robot #100: Handle situation when all waypoints have been reached

Nanoseb, 2016-07-09:
Right now when the last waypoint is reached the tasks node crashes without any further indication:

```
Traceback (most recent call last):
  File "/home/seb/sailing-robot/src/sailing_robot/scripts/tasks", line 40, in <module>
    goal_heading_publisher(tasks_runner)
  File "/home/seb/sailing-robot/src/sailing_robot/scripts/tasks", line 25, in goal_heading_publisher
    state, goal_heading = tasks_runner.calculate_state_and_goal()
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks.py", line 74, in calculate_state_and_goal
    self.start_next_task()
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks.py", line 65, in start_next_task
    self.active_task = self.tasks[self.task_ix]
IndexError: list index out of range
```

takluyver, 2016-07-09:
Thanks, I've been meaning to talk about this but forgetting.

smaria, 2016-07-10:
For testing, it would probably be useful to have a way of looping;
otherwise the node could just inform the ROS operator via logwarn that all waypoints have been reached

takluyver, 2016-07-13:
For now, I have just made it always loop - if there is something we want it to do after the 'last' task, such as returning to base, we should just add that as an extra task at the end.

closed 2016-07-13 by takluyver