Maritime-Robotics-Student-Society/sailing-robot #96: Errors in tasks_ros.py

Nanoseb, 2016-07-09:
I now have issues with the tasks.py node, I have the folliwing error:
```
Traceback (most recent call last):
  File "/home/seb/sailing-robot/src/sailing_robot/scripts/tasks", line 38, in <module>
    tasks_runner = RosTasksRunner(tasks, nav)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 13, in __init__
    super(RosTasksRunner, self).__init__(*args, **kwargs)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks.py", line 41, in __init__
    self.tasks = [self._make_task(d) for d in tasks]
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 56, in _make_task
    self.register_debug_topics(task.debug_topics)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 39, in register_debug_topics
    mod = importlib.import_module(dt_mod)
NameError: global name 'importlib' is not defined
```
I fixed it by adding `import importlib` in the header of the tasks_ros.py file, but then I have:
```
Traceback (most recent call last):
  File "/home/seb/sailing-robot/src/sailing_robot/scripts/tasks", line 38, in <module>
    tasks_runner = RosTasksRunner(tasks, nav)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 14, in __init__
    super(RosTasksRunner, self).__init__(*args, **kwargs)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks.py", line 41, in __init__
    self.tasks = [self._make_task(d) for d in tasks]
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 57, in _make_task
    self.register_debug_topics(task.debug_topics)
  File "/home/seb/sailing-robot/src/sailing_robot/src/sailing_robot/tasks_ros.py", line 41, in register_debug_topics
    dt = getattr(mod, dt_cls)
AttributeError: 'module' object has no attribute '/heading_to_waypoint'
```
And I am not sure how to fix this one.
(ping @takluyver ;) )

takluyver, 2016-07-09:
Sorry, this is just code that I haven't had a chance to test since I changed it. Fixes coming up...

takluyver, 2016-07-09:
Try with the latest commit on master.

Nanoseb, 2016-07-09:
I still have the same error (the second one).

takluyver, 2016-07-09:
Sorry, I overlooked something. Pull and try again...

Nanoseb, 2016-07-09:
Yes! It is ok now!, Thanks

closed 2016-07-09 by Nanoseb