Maritime-Robotics-Student-Society/sailing-robot #97: Find a 2D visualisation tool for position monitoring

Nanoseb, 2016-07-09:
This will be useful mainly for the simulator and the playback of rosbags.
For now I have found:

* Mapviz: https://github.com/swri-robotics/mapviz
* Rviz: http://wiki.ros.org/rviz

Nanoseb, 2016-07-09:
I have added a node to see the boat with rviz.

Nanoseb, 2016-07-09:
In rviz you just have to select "add", and the topic `/debugging_2D_plot` in the `by topic` list.

In the futur I'll probably add waypoints to the map.

smaria, 2016-07-10:
I would have thought the ardupilot stuff belongs in here?

smaria, 2016-07-10:
https://github.com/gareth-cross/rviz_satellite this looks like a useful visualisation for us

Nanoseb, 2016-07-10:
Oh yes, right, I forgot about them. 
Note that the node used right now publish `visualization_msgs` that can be opened by rviz, mapviz and 2D_plot_tool.

Nanoseb, 2016-07-13:
Rviz is working pretty well, I am going to document that on the wiki.

closed 2016-07-13 by Nanoseb