Maritime-Robotics-Student-Society/sailing-robot #98: Check PID values for the rudder (sign issue)

Nanoseb, 2016-07-09:
During my tests and validations with the simulator I noticed that the rudder PID values are reversed (sign problem), the overwritten values during tack are right however. 

I have made an hot fix by setting the P value as an negative number (in the simulator branch), but the problem have to be investigated further.

tsaoyu, 2016-07-09:
I have looked at related nodes, the issue is `pid_rudder` is following error convention of  `PID` class (https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/src/sailing_robot/pid_control.py#L74 ). And the controller is negative feedback x = -Kx (https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/src/sailing_robot/pid_control.py#L62). So I think it should be fine to use negative P value for the controller.

tsaoyu, 2016-07-09:
But negative P is not intuitive at all. So I added a minus sign x = -(-Kx) = Kx to keep both error convention and control command right. 
ref: https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/scripts/pid_rudder#L31

closed 2016-07-11 by Nanoseb