Maritime-Robotics-Student-Society/sailing-robot #106: Jybe when necessary

Nanoseb, 2016-07-13:
The boat is only doing tacks when it wants to change tack even if sometimes a jybe is more appropriate.

takluyver, 2016-07-13:
See also #31 - because we don't currently track where the sail is, we may jibe unexpectedly when going downwind. It's not clear whether this is a problem with no human sailors to be hit by the boom.

Nanoseb, 2016-07-13:
We don't really care about where the sail is and unexpected jibing here I think. Unlike a tack, you can jibe when ever you want and you will not loose speed.

If we want to improve performances of the boat we should avoid going between ~165 and ~195, and implement a 'jibing' strategy similar to the tacking one though.

Nanoseb, 2016-07-13:
[image]
(wind from the south and the boat is going north)

Nanoseb, 2016-08-18:
Just realised that the area scanning challenge will include only jibe, so it will save quite some time if the boat can actually jibe.

Nanoseb, 2016-08-26:
When the waypoint in downwind we can just go in straight line without taking care of the wind direction and the jibe, that is the easiest solution, and it will work quite well I think.

Nanoseb, 2016-08-26:
This is a fix that is working, the boat jibe when expected, @takluyver I let you improve it, maybe you want to keep updating the tackvoting when going downwind, or reset it (?).

takluyver, 2016-09-02:
I've just added a parameter that we can switch it to jibing rather than tacking if conditions make that seem more favourable. Between that and @Nanoseb's fixes, I think we can close this.

closed 2016-09-02 by takluyver