Maritime-Robotics-Student-Society/sailing-robot #107: Catch boat doing stupid, not reaching waypoint

smaria, 2016-07-13:
Some measure of the baot increasing its distance/not decreasing its distance to the waypoint could help prevent the boat getting stuck in an error of our logic.
If a faulty behaviour is detected, it could cause a random movement.

smaria, 2016-08-25:
boat got stuck trying to tack for ages today, whilst at the same time continuously increasing its wp distance. we should make the boat jibe towards its goal heading if it can't get there by tacking within dT, dT ~ 1/windspeed

takluyver, 2016-09-02:
I've just added a parameter that we can switch it to jibing rather than tacking if conditions make that seem more favourable. Between that and @Nanoseb's fixes, I think we can close this.

closed 2016-09-02 by takluyver