Maritime-Robotics-Student-Society/sailing-robot #111: Obstacle avoidance

takluyver, 2016-07-18:
Can we use a webcam to detect the presence of orange blobs in our path?

Nanoseb, 2016-08-18:
The node `sensor_camera_detect` publish in the topic `camera_detection` if anything is in front of the camera (and have the right color (set in the parameter file)). 

The 'calibration' procedure is handled by the file 'utilities/camera_detection_calibration.py' (right now only the threshold can be calibrated with this script, and have to be copied/pasted in the parameter file).

takluyver, 2016-08-19:
Thanks Seb! I imagine that the higher level code using this will look something like:

- Sail between two waypoints repeatedly
- If the obstacle is detected when we're heading towards the waypoint, insert an extra waypoint, which will take us out of the normal course. Once we reach the extra waypoint, continue going to the waypoint we were already heading for.

Nanoseb, 2016-08-19:
Also because we have to stay in the 20m large 'channel' when there is no obstacle, we will probably need more than 2 waypoints (just add some others in between to be sure), but yes the main idea seems good.

takluyver, 2016-09-03:
So we have the detection working (hopefully), but we still need to integrate it to make a decision and sail the course.

[image]

This is my (fairly simple) idea. A, B, D and E in this diagram will be normal waypoints for it to sail to. Then after B (in the left-to-right leg) and D (in the right-to-left leg), there will be a task with conditional waypoints C0 and C1. If while the boat is pointing towards C0 it detects the obstacle, it will switch to head towards C1.

We want to position B and D not too close together, so that it can leave the course and return without hitting the obstacle or tacking. C1 needs to be far enough out of the course that we don't hit the boats making the obstacle, but close enough that we don't spend too much time going to it and back.

Obviously many more sophisticated schemes are possible, but time's pressing ;-)

Nanoseb, 2016-09-03:
ok now everything is ready, just need some testing.

Nanoseb, 2016-09-03:
just tested with the simulator and it is working as expected.

takluyver, 2016-09-03:
:tada:

takluyver, 2016-09-12:
The detection worked in the real challenge - unfortunately it was too late for the boat to turn away, but we definitely detected the obstacle.

closed 2016-09-12 by takluyver