Maritime-Robotics-Student-Society/sailing-robot #112: Test wifi range

takluyver, 2016-07-18:
We were still having issues with the wifi yesterday when the boat was in some parts of the lake. We can test this on the lawn at Boldrewood.

tsaoyu, 2016-07-18:
Compatible list:
Magellan works with Edimax WiFi dongle

Sail_Robot works with ThePiHut WiFi dongle and Edimax high gain WiFi antenna (interchangeable)

https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Wifi-and-access-point-on-RPi

I will remove all the AP setting on Magellan to and stop to support Edimax dongle.

takluyver, 2016-07-18:
We failed to get the big wifi dongle working yesterday, so we resorted to the small one, but we placed it on top of the boat.

tsaoyu, 2016-07-18:
So it worth to try that antenna at next time.

takluyver, 2016-09-12:
We tested the wifi range in Viana and got at least ~200m. However we were still having wifi issues (#139), whether because of range or interference or some combination.

closed 2016-09-12 by takluyver