Maritime-Robotics-Student-Society/sailing-robot #113: Keep GPS running

takluyver, 2016-07-18:
The GPS seems to go to sleep if nothing's reading from it (probably a power saving measure). This is annoying as it seems to have trouble getting a signal initially when it wakes up. We want to find some way of keeping it running.

Nanoseb, 2016-07-18:
Maybe just `cat /dev/ttyACM0 > /dev/null` will do the trick.
Or a cron job every x minutes with `dd if=/dev/ttyACM0 if=/dev/null bs=1M count=1`

takluyver, 2016-07-18:
I suspect that if we have something else trying to read from it, it will cause issues for our code actually trying to use it, because each of them will read random chunks of the data. If that's the case, we need something a bit more complex to read and forward the data. Sophia suggested keeping ROS plus the GPS node always running, but I'm not sure about that - I think it's useful to be able to completely shut down and restart the ROS machinery.

I should have mentioned that we do have a workaround: the GPS appears to stay on for some time (seconds, maybe a minute?) after it's read, so if we shutdown ROS and immediately start it again, we still have a fix.

Nanoseb, 2016-07-18:
We can make the cron job slightly smarter, and check at every run if the gps node is running.

smaria, 2016-08-20:
From ublox datasheet:
Time To First Fix
Cold Start: 29 s
Warm Start: 28 s
Hot Start: 1s

This document has more details on the power save modes: https://www.u-blox.com/sites/default/files/products/documents/PowerManagement_AppNote_%28UBX-13005162%29.pdf
My housemate has a windows PC and suitable adapters to change the GPS mode, I will have a talk with him.

For now it sounds to me like we can use the fix of just keeping roscore + the gps node running, and just killing roscore to restart it directly after.
We also don't need GPS running the entire time, only when we switch between different tests.

One thing to note when keeping roscore running: Parameters are kept. This means, we might get outdated parameters hanging around if we don't set them to a new value in the launch script. I don't think this is an issue.

Conclusion: Make separate always-on launch script. Talk to housemate about changing GPS settings.

takluyver, 2017-09-10:
Closing for now because this hasn't been a big problem in practice.

closed 2017-09-10 by takluyver