Maritime-Robotics-Student-Society/sailing-robot #116: Soft limits on sail servo

tsaoyu, 2016-07-18:
Because our servo works as continuous rotate motor at lower and upper PWM range it is necessary to set these two soft limits on lowest level node.  Test scripts `servo_key.py` can be found in utilities folder. One can change the lower and upper limits and use keyboard to find out these limits.  

https://github.com/Maritime-Robotics-Student-Society/sailing-robot/blob/master/src/sailing_robot/scripts/actuator_driver_sail

smaria, 2016-08-20:
Sorry, what is the issue here? This looks like docs to me, which should then go in the boat setup in the wiki.
E.g. here:
https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Pre-sail-Raspi-setup

tsaoyu, 2016-09-12:
Done by self test scripts now.

closed 2016-09-12 by tsaoyu