Maritime-Robotics-Student-Society/sailing-robot #120: Can't tack in moderate waves

takluyver, 2016-08-06:
We tested today at the upper end of the wind speeds we may have to sail in, and there were small waves (maybe 10-20cm). We discovered that it was almost impossible to tack in these conditions, even with manual control - the boat being very light, it has little momentum, and as soon as it tries to turn into the wind, it loses speed.

We were still able to jibe, so we should probably include an option to jibe rather than trying to tack for these conditions.

The boat was also not able to sail very close to the wind, so we would have had a very hard time going upwind.

pierbiag, 2016-08-08:
All these problems related to overhealing and maneuverability in strong winds will ideally be solved by the smaller set of sails we are purchasing 

Pier Maria Biagiolini

takluyver, 2016-08-08:
Great. Do we know when they'll arrive?

pierbiag, 2016-08-08:
I'm waiting a reply from the guy at sails etc so we can confirm the purchase

smaria, 2016-08-20:
Sails are here. However, jibing is still something we need to teach our boat to do.

smaria, 2016-08-20:
#106 deals with jibing. it may be worth adding to jibe when tacking hasn't worked a couple of times at a later time.

takluyver, 2016-09-02:
Closing this as we have both smaller sails and code for jibing. If we're tagging issues 'wontfix', we should probably close them.

closed 2016-09-02 by takluyver