Maritime-Robotics-Student-Society/sailing-robot #122: Water got in

takluyver, 2016-08-06:
Some water got in the boat today, and we lost control with the remote. @NanaOKA's expert paddleboarding saved the day, but we should try to waterproof it better.

Can we also design the servo control to do something useful if it fails - e.g. maybe let the sails all the way out, so that the boat slows down.

tsaoyu, 2016-08-06:
[image]

smaria, 2016-08-18:
How does the boat determine if it is a connection loss - or just a case of 'sailing as usually', remote switched off?

It would be possible to connect one of the receiver pins to the RPi directly, e.g. another pin from a switch that we always leave on. Then if this pin is not set anymore, we would see that on the RPi.

I would _never_ use this in the competition, however in the long run this can be used to then make the boat sail to a fixed home position. Until wrsc this imho has low priority.

takluyver, 2016-08-19:
I think we may have had water in the compartment containing the radio receiver; if that's the case, I suspect there's little we can do with the high-level control to rescue it. It was more a vague hope that the physical systems could be arranged so that a catastrophic failure results in it stopping rather than sailing away.

I agree that this is low priority, though - the safety boat for the contest should be able to catch it if it runs amok.

smaria, 2016-08-20:
we are currently waterproofing various bits, this includes the holes in the receiver compartment, which should fix this - unless there is leakage through the lid.

takluyver, 2016-09-02:
We hope that the improved waterproofing has fixed this.

closed 2016-09-02 by takluyver