Maritime-Robotics-Student-Society/sailing-robot #123: Can't use rviz with maps on the project laptop

takluyver, 2016-08-06:
Rviz complained the the aerial maps plugin couldn't be loaded. I guess the package needs to be installed on that laptop.

Nanoseb, 2016-08-07:
Yes you just have to follow the wiki page: https://github.com/Maritime-Robotics-Student-Society/sailing-robot/wiki/Rviz:-2D-plot-of-our-boat

takluyver, 2016-08-08:
Thank you! On Wednesday, let's do that on the project laptop, check that it's working, and then close this.

Nanoseb, 2016-08-08:
Also since you probably don't have a data connection during lake test, you should run a simulation with the lake waypoints beforehand for rviz to download map tiles. Then you'll be able to use it offline (you can always backup the folder `~/sailing-robot/src/rviz_satellite/mapscache` somewhere else in case the cache gets deleted).

takluyver, 2016-08-11:
I have followed the instructions from the wiki and got the plugin loading, but it doesn't show anything. There's a warning saying "No map received". But I'm also not sure if I'm using ROS right,  because all my `rostopic echo` commands aren't working against the rosbags.

Nanoseb, 2016-08-11:
Is rviz working correctly during simulations ? And for the map to show you need an internet connection so if you are connected to the pi maybe it is not possible (you'll need to run a simulation beforehand to dl maps to have them in cache)

takluyver, 2016-08-11:
I was trying to replay a ros bag file, and I was connected to the internet.
I'm not there next week, but hopefully someone else can test rviz with the
simulator on that laptop.

Nanoseb, 2016-08-11:
I have never tested rosbag playback with rviz, I'll give it a try.