Maritime-Robotics-Student-Society/sailing-robot #124: Boom dragging in the water

takluyver, 2016-08-06:
Because the boom is quite low, even modest healing today made it touch the water as waves came past. I don't know if there's anything we can do about this, but it presumably slows the boat and causes it to turn away from the wind.

Nanoseb, 2016-08-07:
I don't think there is anything to do, sheet in will increase the heeling, maybe sheet out to heel less, but then we will also decrease the boat speed (it will just be more steady). 

Or maybe by facing the wind a little bit more, waves are going to come less from the side and be less a problem, but again it will slow down the boat.

smaria, 2016-08-18:
We now have smaller sails for strong winds, so this should not be an issue anymore.

closed 2016-08-18 by smaria