Maritime-Robotics-Student-Society/sailing-robot #126: Real north and magnetic north

Nanoseb, 2016-08-12:
The north given by a compass is not the real one, so an offset have to be added to the magnetometer values to account for this difference. This offset depends on the location and the date (year). 

I think we can hard code that or use a parameter (it will have to change between England and Portugal). The difference should be less than 8°, but still that is a correction easy to implement.

The correction should be made directly in the `sensor_driver_imu` code since the value without the correction is not useful.

takluyver, 2016-08-12:
Thanks Seb. I think doing this as a parameter is the best way.

According to [this site](http://www.ngdc.noaa.gov/geomag-web/#declination), the current declination is:

- Southampton: 1° 1' W 
- Viana do Castelo: 2° 36' W

I think those numbers refer to where magnetic north points relative to true north, so we need to apply them in the other direction to find true north. So we should be adding a positive value to the compass reading in both cases.

Nanoseb, 2016-08-12:
Yes, perfect, I thought the correction was larger than that, but still 2.5° is something.

Nanoseb, 2016-08-31:
I added a parameter (that will have to be changed in Portugal !) to offset the heading.

Nanoseb, 2016-08-31:
By the way, are you sure we have to add positive values ? I think I always learned that from the magnetic north to the true one (in europe) we have to subtract a quantity.

Also in europe we have a west declination, and according to [wikipedia](https://en.wikipedia.org/wiki/Magnetic_declination#Non-adjustable_compasses) it means adding a negative value to a compass to have the true north.
Can you double check that also to be sure.

takluyver, 2016-08-31:
Yep, sorry, thinking it through again, you're right - westerly declination means the magnetic bearing we take will be a couple of degrees too high, so we need to subtract it (/add a negative). I think it's best to just make the numbers in the parameter file negative, as that fits with the usual east/west convention.

commit 90b96b19 (2016-08-31): fix north declination values

close #126
closed 2016-08-31 by Nanoseb