Maritime-Robotics-Student-Society/sailing-robot #133: add a (MIT) licence to our code, so people can actually use it

smaria, 2016-08-21:
I had some conversations with nanoseb and thomas about adding a licence to our repo. More specifically, the MIT licence [0]. To get ready for adding the licence, everyone please mark code you mainly worked on 'READY FOR MIT LICENCE' if this code is yours/written by sailing robot team members or an according licence text based on its sources.
Ask for help if you are uncertain. I believe Sebastien is looking into finding a licence for the IMU code.

[0] The choice was between a BSD/MIT style open for anything licence and a GPL style licence that would enforce all projects using our code to use the same licence. The core of ROS is BSD licence, and there were no strong arguments for using GPL instead. MIT is more popular on github and simpler than BSD (since there are there generations of BSD but just one generation of MIT), so we went for that.
If someone has concerns with this choice please comment/let me know in one of our next meetings.

takluyver, 2016-08-21:
Rather than trying to work out who mainly worked on which bit of code and add comments to it all, can I suggest a simpler way of doing this:

Anyone who has contributed code, you have a week from today to express any reservations about using the MIT license for it. If no-one objects, we will assume we're all happy with that, and apply the MIT license to the whole project (except any bits of code we have copied from other sources - we should try to identify and credit those).

Nanoseb, 2016-08-22:
We just got an answer from the writer of the imu driver code: https://forum.pololu.com/t/lsm303d-raspberry-pi-driver/7698/6
So we can use it without limitations.

tsaoyu, 2016-08-23:
`pigpio`: free and unencumbered software https://github.com/joan2937/pigpio/blob/master/UNLICENCE 

`Python pid ` BSD License 

`ros_lib` ROS Arduino library BSD License

smaria, 2017-02-11:
The licence was added

closed 2017-02-11 by smaria