Maritime-Robotics-Student-Society/sailing-robot #135: Remove Tack voting when the boat round a buoy

Nanoseb, 2016-08-23:
Right now the tack voting system is enabled when the boat is close to a waypoint, however to round a buoy we need to have several waypoints (2 or 3) close to the buoy, thus close to each other. Between them we don't want the tack voting system because we need to be able to tack without having to wait 7.5 seconds.

Nanoseb, 2016-08-24:
I think we will need to have 2 'type' of waypoints, 
- the first one can be considered like usual waypoints (with the tack vote system when we get closer to it) and can be used for the first wp around the buoy.
- the second the rest of wp around the buoy, when heading to these one the boat need to be able to tack/jibe without waiting

I think the easiest way of implementing it without breaking compatibility with previous parameter file is to add a list of wp of 'type 2' in the waypoint parameter file, something like
`wp/list_no_vote: ['2', '5', '8']`

Nanoseb, 2016-08-24:
according to the rules we don't need to round buoy...

> If a boat succeeds completing the whole course, it will be scored by its arrival order. If not,
boats will be scored according to the number of marks reached in the correct order and the
order of arrival to that mark. [...] A mark/buoy is considered reached if at least
one track point is recorded within a radius of 5m around the position of the (virtual) buoy.

So in the end this is not needed.

closed 2016-08-24 by Nanoseb

takluyver, 2016-08-24:
Nice, that makes the course sailing quite a bit simpler; especially as 5m
is a more generous tolerance than we've been aiming for.

Nanoseb, 2016-08-24:
Moreover the 5m tolerance is based on our gps data (the log file we provide to them), so no worries on that point either.

takluyver, 2016-08-24:
Hmmm, not sure about that. I think they might refer to their gps data as well.

Nanoseb, 2016-08-24:
> As the scoring process will be based on the course done by each boat, it will be preferable to use the actual position data processed by the boat’s navigation system to prevent possible deviations resulting from the different location data produced by the trackers and the boat’s navigation system.

takluyver, 2016-08-24:
OK, great. I might still set a 4m target zone just to be on the safe side.