Maritime-Robotics-Student-Society/sailing-robot #137: Always jibe parameter is not working properly

Nanoseb, 2016-09-02:
I have tested the parameter `jibe_to_turn` in the simulator, in this case it worked once but on the second jibe it starts making 360° and after 3-4 of them the boat goes in the exact opposite direction to the waypoint.

And sometime it is even without making 360° but the boat directly goes in the opposite direction.

Nanoseb, 2016-09-02:
Some more info, when the boat goes away from the waypoint its sailing_state is `normal` and during 360° it is `jibe_to_port_tack` the entire time.

Here the boat starts to make loops, the wind is coming from the north and the waypoint is the one on the top right.
[image]

Wind conditions are the same, and the boat didn't make any 360 here.
[image]

Also sometimes the boat also just keeps making 360.

Nanoseb, 2016-09-02:
I have tried some things but same result...

Anyway I have tried my 'force_jibe' node that is making the boat jibe when a tack is failing (= take too much time), and this one is working well so maybe the always jibe is not entirely needed.

takluyver, 2016-09-03:
I've tracked down the going round in circles issue - I'd missed out the `el` of an `elif`. I'm less sure about what makes it head off in the wrong direction.

@Nanoseb rviz doesn't work on my ROS VM (it segfaults) - do you have time to test this again, since you can see better than me what's going on?

Nanoseb, 2016-09-03:
Yes, I can test this right now.

takluyver, 2016-09-03:
Thanks! :smiley:

Nanoseb, 2016-09-03:
Ok it seems to work correctly now!

takluyver, 2016-09-03:
Excellent. Those two characters make all the difference :-)

closed 2016-09-03 by takluyver

Nanoseb, 2016-09-03:
Can you make it so that it will try to jibe only when a certain topic is published ? (so we can detect if a tack is failing)

takluyver, 2016-09-03:
I'm not exactly sure what you're thinking. I'm trying to work on #51 now, so if you can see what you need to do, go for it.