Maritime-Robotics-Student-Society/sailing-robot #139: Wi-fi connection issues

takluyver, 2016-09-12:
Having a wifi connection to the boat makes it very easy to get full SSH access and change anything remotely. But during the competition we repeatedly had problems establishing and maintaining a wifi connection from the shore.

Ideas, in rough order of increasing effort:

1. Continue to live with limited connectivity from the shore, so the people in the RIB have primary responsibility for control. However, we've had occasions when the shore team has identified and fixed a problem (e.g. the need to git pull on the raspi) which the RIB team may have missed.
2. Upgrade wifi antennas, on the boat and/or on land. I think the connection was a bit better with the bigger antenna on the boat, but it was still an issue. Even with big antennas, the problem may partly be interference, especially if other teams are using wifi connections.
3. Mobile data connection on the boat; designed for greater range, moving systems and small antennas. However, latency will certainly be greater, and bandwidth may be small, depending on the area. We'd need to sort out local SIMs, but maybe could coordinate that with competition organisers, since I think they use SIMs for the GPS trackers.
4. Other point-to-point radios, such as XBee. These are more designed for the kind of range we want, but it would be more work to integrate them to do useful things.

TAJD, 2016-10-07:
I looked into bridging the ship/shore comms gap briefly at the end of my GDP. As a result I've had a few thoughts on the matter.

1. I'm guessing this works and has been proven as a reliable method (whilst frustrating and not ideal)
1. I'm not sure even upgraded WiFi antennas have the range required. It might be ideal to exert control over the range of 100s of meters from shore and I don't think WiFi has the range to do this. Plus larger antennas would add weight and possibly more points for water ingress to occur. 
1. I've got a sim shield to plug into an Arduino, I've not tried using it yet though. The issues with this method are that sim shield itself is relatively bulky and that the sim data would need to be constantly topped up - this is spending money and it would be good to minimise that.
1. I've not used xBees themselves, and whilst they are relatively expensive they seem to be widely used and extremely good at communication over the sort of distances we're looking at. It might need a little bit of work to set up, but the investment of time depends on how critical we think a reliable ship/shore link is. 

This comms link looks like a really key issue, but to identify a good solution we'd need to look at how we're willing to invest time/resources as well as the physical space on the boat and how much time we'd need to spend integrating the components together. I'd be quite interested in sorting out xbee communication as they're small, have good range and are low power. However they are relatively expensive.

pierbiag, 2016-10-07:
There are similar "plug and play" solutions for serial interface from rs (lora range) that are much cheaper and apparently more efficient. However they are already investigating the matter if we can do something useful (ssh-like) over such a low bandwidth link.

Pier Maria Biagiolini

takluyver, 2016-10-07:
As to the importance: I think there were relatively few occasions where the shore link actually made a difference. But it is nice to track it live, especially for demonstrating the boat to other people. So if we have the resources to pursue improvements, I think it's worth playing with.

takluyver, 2017-09-08:
We actually went backwards on this for 2017. :-( The wifi antenna is now with the raspi, rather than at the top of the mast, and so we get very little range on the signal, especially when there are waves. This was a problem at times, because it wasn't always clear whether ROS was still working, or when the boat had finished a challenge.

I think this is a big improvement that could be made for next year, whether it's a full control interface like we have with wifi, or a limited monitoring interface to have some idea what's going on.

tsaoyu, 2017-09-08:
https://haxiot.com/shop/lorawan-usb-client-us900/?doing_wp_cron=1504859862.1429119110107421875000

takluyver, 2017-09-08:
That looks interesting, though from a quick scan of the documentation it looks like more work than I had thought. I thought LoRa would give us a network interface which we could use like wifi, whereas the docs describe it as a serial interface which you send commands and data to. I'm sure we could make use of that, but it would probably be more work for less control than we have with wifi.

takluyver, 2017-09-14:
https://homes.cs.washington.edu/~gshyam/Papers/loRaBackscatter.pdf

(The system they describe is in the research stage, but it's encouraging that people are doing cool things with LoRa)

takluyver, 2017-11-18:
Suggestion from @griz1112 on issue #142:

> Use a RFD900x telemetry radio with mavlink. Has two channels and unreal range 10km with stock antenna up to 40km with upgraded antenna and directional pointing.

- [Example RFD900x modem](https://www.helibatics.com/rfdesign-rfd900x-modem-rc-control-and-telemetry-radio-modem-with-diversity/)
- [Mavlink ROS package](https://github.com/mavlink/mavros/blob/master/mavros/README.md)
- It sounds like Mavlink is designed for air vehicles - there may be some assumptions baked into it that don't hold for our use.

tsaoyu, 2017-12-05:
Considering get a pair of [USB LoRa](https://yadom.eu/reseaux-iot/lorawan/carte-de-communication-lorawan-usb.html). 

- It use USB serial interface, leave physical UART pins for GPS module and easy to plug into laptop
- Have SMA connector, would better with RP connector and use [RP<>SMA](https://yadom.eu/modules/radio-modules/cable-ufl-rp-sma-100mm.html) 
- access terminal using PPP over UART (bandwidth is limited but good enough to launch the system)

griz1112, 2018-01-17:
You can use mavlink for anything. Ardupilot has versions for planes drones boats even submarines. I have it on my rover. If you are coming from a ROS env use mavros to integrate the two.