Maritime-Robotics-Student-Society/sailing-robot #140: Human communications

takluyver, 2016-09-12:
The 'vi in a screen' chat system was an inspired hack but not ideal:

- We all share one buffer, so if two people type at once, you get a mess of mingled characters.
- The only way to tell if you're still connected is to try changing something
- It's easy to forget about it, so questions readily go unanswered.

Ideas:

1. Dedicated chat system over local wifi - are there existing solutions? How well do they handle flaky connections (see #139)?
2. Walkie-talkies - but this requires an extra hand to use.
3. Audio headsets, as used by gamers to communicate without requiring hands.

tsaoyu, 2016-09-12:
After some search on GitHub 
https://github.com/shazow/ssh-chat

and Google shopping

http://twoway-radioshop.co.uk/motorola-tlkr-t40-radios.html?gclid=Cj0KEQjwpNm-BRCJ3rDNmOuKi9IBEiQAlzDJH4ShG8Ja47YXBl5HW79QPpBdhZpDOyBmcu4Q4cCZiacaAjjd8P8HAQ

and

http://www.onedirect.co.uk/motorola-tlkr-t80-2-x-closed-face-helmet-mics

takluyver, 2016-09-12:
I've also seen ssh-chat, but I'm not sure if it's what we want for this. E.g. I don't think it stores messages at all, so if you say something while I can't connect, I never see that.

Nanoseb, 2016-09-12:
apparently there is some sort of history, try the example server `ssh chat.shazow.net`.

takluyver, 2016-09-12:
True, so maybe that would work.

smaria, 2016-09-24:
still needs us connected to the boat network, only works for people with a laptop etc. I am a big fan of the walkie talkies, they will help so much just on normal lake tests.

pierbiag, 2016-09-24:
I should have a couple somewhere, I'll see if they still work and in case bring them 

Pier Maria Biagiolini

tsaoyu, 2017-12-12:
Got a pair of walkie talkie with audio headset.

closed 2017-12-12 by tsaoyu