Maritime-Robotics-Student-Society/sailing-robot #141: See when manual remote control is active

takluyver, 2016-09-12:
Can we tell from the Raspi which state the multiplexer is in? It would be really useful to display this in the dashboard and record it in ROS bags.

tsaoyu, 2016-09-12:
We have an extra wire reserved between RPi and Multiplexer for this purpose. 
An easy way is to pass selection signal on channel 5  to RPi.
On software side, use `pigpio` read to monitor http://abyz.co.uk/rpi/pigpio/python.html#read .

takluyver, 2016-09-12:
Sounds like this should be a matter of one simple ROS node to turn the GPIO state into ROS messages, then :-)

Nanoseb, 2016-09-15:
The code also have to handle when the remote is off. The multiplexer have a failure state for that to switch to autonomous when no signal is received. This doesn't need to be a 3rd state but need to be seen as autonomous mode.

takluyver, 2016-09-15:
I'm hoping that the multiplexer will show us the same thing for that case as when the RC is on but with autonomous control enabled. But this will all have to be tested when we've got the multiplexer.

Nanoseb, 2016-09-15:
The idea was to connect the raspberry pi to the receiver pin, so I am not really sure about that, but yes anyway this will have to be tested with the multiplexer.

takluyver, 2017-09-08:
This is working nicely.

closed 2017-09-08 by takluyver