Maritime-Robotics-Student-Society/sailing-robot #142: New ground station and mission control system

tsaoyu, 2016-09-12:
We have a nice web-based light ground station system to monitor our boat. It is good enough for simple tasks and missions in this competition. 

In a long run, we may need:
1. On server side, better graphical interface on with our existing web-based monitoring page
2. On client side, find suitable mission control system which can display more information at same time
https://github.com/nasa/openmct

http://wiki.ros.org/ground_station

mltomb, 2016-09-15:
Hello

I found our github few days ago. It is cool to be able to see this type of discussion.

We tried this ground station this year.
http://ardupilot.org/planner/docs/mission-planner-overview.html

It was really useful. It can understand multiple protocols in network or serial link. We used mavlink protocol trough wifi.

Nanoseb, 2016-09-15:
Thanks for the tip, we tried to use it with ROS some months ago (there is a package to convert ros messages to mavlink) but we had some issues (don't remember what). I think we can give it an other try, having a live map is quite useful. Right now we are using rviz (a ROS ploter, but it is quite basic and not perfectly suited for that).

tsaoyu, 2016-09-15:
mltomb, You are welcomed to join our conversation! 

We have investigated the possibility in using APM planner as our ground station system. The problem is that we don't have a reliable wireless connection for MavLink. At that moment, we only use a tiny USB dongle for wireless communication. 

New we have a more powerful WiFi antenna, it is time for to have another try. 

This problem is related to https://github.com/Maritime-Robotics-Student-Society/sailing-robot/issues/139

griz1112, 2017-11-12:
Use a RFD900x telemetry radio with mavlink. Has two channels and unreal range 10km with stock antenna up to 40km with upgraded antenna and directional pointing.

takluyver, 2017-11-18:
Thanks, I copied the suggestion to #139,  which is more about the radio link.