Maritime-Robotics-Student-Society/sailing-robot #143: Next plan for our website

tsaoyu, 2016-09-12:
We are using a second level domain with free WordPress subscription. It has a fairly easy to use editor and control panel. But it also have many restriction. For example, we are not able to put our crowd funding widget on the page.

 And to better publicise ourself, a designated domain like `www.sailrobot.co.uk` or `www.sailrobots.com` would be great. 

Also, our WordPress is the only domain we used in print material. To keep that, options including:
* Keep using WordPress, upgrade to paid plan
* Redirect to new domain (£11/year)
* Buy a new domain and redirect to WordPress page

If we abandon WordPress system we can:
* Use GitHub pages
* Rent our own VPS e.g. DigitalOcean

takluyver, 2016-09-13:
For reference, the Wordpress $3 plan includes a custom domain. Domains cheaper than redirects: lazy way to keep people around. Their registration [doesn't include](https://en.support.wordpress.com/domains/#available-domains) `.uk` domains, though.

I would go for a domain like sotonsailbot, and end with either a UK suffix or a `.org`.

If we move away from hosted solutions, I'd rather go for a static site hosted on Github pages, so we don't need to run our own server.

tsaoyu, 2016-09-21:
Aber team launched their web page with [GitHub page](abersailbot.github.io).
It can be a good reference for us.

SuperWool, 2016-09-23:
`sotonsailbot` sounds good. Do we go for that?

tsaoyu, 2016-10-07:
`www.sotonsailrobot.org` is up and running now

closed 2016-10-07 by tsaoyu

takluyver, 2016-10-07:
And it even has a picture of us rather than a stock photo now! :-)