Maritime-Robotics-Student-Society/sailing-robot #144: X86 micro-controller?

tsaoyu, 2016-09-13:
https://software.intel.com/en-us/iot/hardware/joule

Key-features:

* 64bit X86 arch 
* ROS compatible
* 48GPIO with 4 hardware PWM 
* MHF4 connectors for external antennae
* on chip Intel HD graphics card

takluyver, 2016-09-13:
Jealous of the boat running Windows? ;-)

I don't think we ran into any limits of computational power or connectivity, so I'd be inclined to stick with the Raspi - they're cheap, readily available, and there's a lot of information already out there from people who've already done all kinds of things with them. The most attractive thing on the feature list is the connections for antennae, but I don't know if we'd actually use that.

tsaoyu, 2016-09-13:
Haha, I am really Jealous on their Windows 10 computer stick.

The most attractive part for me is X86 arch with Ubuntu Core. 
It opens greater possibility for more ROS libraries. 
For example, sensor fusion library with http://wiki.ros.org/robot_localization 

It also have drawback like smaller user community and higher price.

takluyver, 2016-09-13:
I'm confused - does that library require x86? Or does it need more processing power than we have?

Nanoseb, 2016-09-13:
The processing power is really not an issue atm, when every thing is launched we use about 10-20% of the total cpu, even the opencv lib did not use much cpu

tsaoyu, 2016-09-13:
No, this library requires dependencies which is not available on ARM.

takluyver, 2016-09-13:
Which deps are those? Searches aren't turning up much info.

tsaoyu, 2016-09-13:
Don't remember atm, I  will let you know once I meet Brain

 [image]

takluyver, 2017-09-10:
Closing for now as we have no current plans to move to x86. We'll keep our eyes open for different options, though.

closed 2017-09-10 by takluyver