Maritime-Robotics-Student-Society/sailing-robot #145: exterior LCD display on board

tsaoyu, 2016-09-13:
[image]

A display like this with information like:
* Name of current launch file
* Power level of battery
* Autonomous/Manual mode

Nanoseb, 2016-09-13:
Haha, the launch file name, good idea ;)

smaria, 2016-09-24:
This looks like it will be a mess during waterproofing and will very likely be covered up by bumblebee tape in most use cases. How about we display the name of the launch file on the dashboard? (same goes for battery state, once we can read it)

takluyver, 2016-09-24:
I agree that info on the dashboard is more practical use in most cases, and less of a concern for waterproofing.

Do we know of a way to get the name of the launch file from inside ROS? My searches for that are not turning up anything.