Maritime-Robotics-Student-Society/sailing-robot #146: More hardware support

tsaoyu, 2016-09-14:
To make more people use our code on their own boat. 
We are planning to extend our hardware support.
Raspberry Pi family would be a good start:
*  **Raspberry Pi 3**
* Raspberry Zero

and another popular linux single board computer like:
* **BeagleBone Black**
* Odroid  XU4 (http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143452239825)

 **Bold**  indicates micro-computer we now have access to

Nanoseb, 2016-09-14:
I think we first have to focus on documenting what is already working (both the code and the hardware).

tsaoyu, 2016-09-14:
That is also a good point.

takluyver, 2016-09-14:
I also wouldn't focus too much on code reuse until there are people interested in it. A lot of the fun in this kind of thing is building your own systems, so I doubt other teams like us would take our code as a whole for their own boat. They're more likely to use pieces of our code in their own project, but it's hard to anticipate which pieces and what would be different about the context. It's much easier to work this stuff out when someone is actually *trying* to use it.

I agree that we should better document what we have (almost all open source code could have better docs). We should also make sure we can easily hear from likely users of the code (e.g. on the microtransat mailing list).

Nanoseb, 2016-09-14:
Also you will probably recruit (a lot of) new people this year, so a good documentation is needed at least for them.

tsaoyu, 2016-09-14:
Indeed, our Wiki and documentation system is bit messy. And there are two things looks interesting one is nice document done by Aberystwyth team http://boatd.readthedocs.io/en/latest/ , another is the best practice in developing ROS packages http://wiki.ros.org/Get%20Involved .

takluyver, 2016-09-14:
The Boatd docs use Sphinx, which I'm pretty familiar with. That's a fairly manual style of documentation; we have experimented with autogenerating docs from our code, which is less effort than manual documentation but also produces less nice results.

It's only worth making such nice docs once you have a pretty stable API, though - if you're still changing interfaces a lot, then either you spend a lot of time changing the docs as well, or your docs get out of date. Out of date docs are worse than not having docs at all, but they confuse and distract people.

I'd start with the high level description of the system, like the node diagram - which is almost surely out of date, because we've been changing stuff fast and have had higher priorities than redrawing it.

smaria, 2016-09-24:
I put the auto generated docs in a separate github repo, so people don't need rosdocs to see them (but we still keep our code repo clean). However we are trailing off the original topic of this ticket.

Since we would like to switch to the Pi zero anyways, we should focus on that one; and I think nanoseb is already working on that one.

smaria, 2016-09-24:
Sorry, I thought we had a 2xpi2 and 1xpiZero, not a pi3. From what I recall, more people were interested in getting the pi zero to work. I also find it more interesting since it is cheaper, smaller, and computational power should be sufficient as mentioned above.

Nanoseb, 2016-11-22:
I have tested the pi zero over this week end, in the end, it can't really be used like our pi 2 because it is not powerful enough. It can't keep up the rate of 10Hz that we have at the moment, and image processing (for the obstacle avoidance) is taking a lot of cpu as well. 
So it can't replace entirely our pi 2/3.

tsaoyu, 2016-11-23:
Seb, thanks for your report! 
RPi zero seems couldn't meet the computation power requirement, we will not look at this platform until next hardware version released. 

I think we will all agree that RPi 2 is board "officially" supported. For collaborators, we will recommend them use RPi 2 for the best compatibility. (I am still cruise about the compatibility of RPi3.)