Maritime-Robotics-Student-Society/sailing-robot #148: Investigate what happened with the USB cable

smaria, 2016-09-24:
The usb cable that is wired through the boat has a short between at least two wires. What happened? Do we need to improve something about our wiring?

smaria, 2017-02-11:
The USB cable is now fine, aside from some corrosion because it got wet when we were not using it at the competition.
This makes me assume that the usb cable must have gotten water inside, which made the connection.

Conclusion: We cannot assume that USB wires are waterproof.

closed 2017-02-11 by smaria

Nanoseb, 2017-03-04:
Cat6 wires (for outdoor uses): http://www.cabling4less.co.uk/category.php?terms=UT017/50&source=googleproducts&gclid=CjwKEAiA6OnFBRDcgt7YmPKI33ESJACJoTJYgCcpg6Xyx3e5fjLSBkq0Az5LJM2rEYHm8GV7QM2O2BoCcHTw_wcB

posecap, 2017-03-04:
http://uk.rs-online.com/web/p/usb-cable-assemblies/0252062/

Nanoseb, 2017-03-05:
Sorry, I shouldn't have wrote my comment here, it is a closed issue. @posecap you can add your link to the wishlist here: https://github.com/orgs/Maritime-Robotics-Student-Society/projects/2