Maritime-Robotics-Student-Society/sailing-robot #149: Maximise number of computers we can use with ROS, rosbag etc.

smaria, 2016-09-24:
We want to run as many tasks as possible for students to play with ROS. Since we cannot assume they will all have everything set up on their machines from the start, we should have as many machines as possible available for them.

takluyver, 2016-09-24:
Hopefully rviz segfaulting is something peculiar to my VM, and not a general bug in the ROS VM image!

Nanoseb, 2016-09-24:
By changing the port ros uses one can launch several instances of ROS on the same computer, maybe this can be used (see http://answers.ros.org/question/11395/can-multiple-instances-of-gazebo-andor-roscore-run-in-parallel ).
One computer has several copies of the repo and several roscore running, then people can ssh to it to run their tests (however to run rviz in this context one will have to use x forwarding).

smaria, 2016-09-24:
we need to test this with tabarly on thursday. this looks like an excellent plan!

Nanoseb, 2016-09-25:
and the same script using the mac address instead of the ip to be able to keep it even after disconnection of the network. However this will not work properly if there is any router between the server and the client.
```
ip=$(awk '{print $1}' <<< $SSH_CLIENT)

if [ -z "$SSH_CLIENT"  ] || [ "$ip" = "127.0.0.1" ]
then
  port=11311
else
  ip_hash=$(sed 's/\.//g' <<< $ip)
  mac_hash=$(arp -a | grep "$ip" | awk '{print $4}' | sed 's/://g;s/[a-z]/1/g;s/^0*//g')
  port_ip=$((11312 + (ip_hash % 2000)))
  port_mac=$((11312 + (mac_hash % 2000)))
fi
echo ROS port ip: $port_ip
echo ROS port mac: $port_mac
export ROS_MASTER_URI=http://127.0.0.1:$port_mac/

```

closed 2017-02-07 by tsaoyu